ARMmbed/mbed-os-example-client #355: WiFi connection witt client compiled with IAR fails

SamiLappalainenArm, 2017-12-14:
### Description
mbed-os-5.7.0-oob client build by IAR does not register succecfully. Succesfull registering works when GCC_ARM is used to build.

---------------------------------------------------------------
## Bug
Used git repo = mbed-os-example-client => branch mbed-os-5.7.0-oob

Used environment:
- Windows10 laptop 
- K64F + ESP8266
- mbed version 1.2.0
- IAR version 7.80.1.11973

**Expected Behavior**
client shows "Registered object successfully!"

**Actual Behavior**
client shows "onnection to Network Failed - exiting application..."

**Steps to Reproduce**
1) compile using GCC_ARM from CLI using wifi_esp8266_v4.json configuration
2) upload .bin to K64F
3) restart K64F
    Starting mbed Client example
    [EasyConnect] IPv4 mode
    [EasyConnect] Using WiFi (ESP8266)
    [EasyConnect] Connecting to WiFi Anfield-tower
    [EasyConnect] Connected to Network successfully
    [EasyConnect] MAC address 5c:cf:7f:04:6a:f2
    [EasyConnect] IP address 192.168.43.142

    SOCKET_MODE : TCP
    Connecting to coap://api.connector.mbed.com:5684

    Registered object successfully!
4) mbed export -i IAR
    C:\siilo\git\OOB\mbed-os-example-client>mbed export -i IAR
    [mbed] Detected "K64F" connected to "E:" and using com port "COM6"
    Scan: .
    Scan: FEATURE_LWIP
    Scan: FEATURE_COMMON_PAL
    Scan: FEATURE_STORAGE
5) open workspace in IAR
6) rebuild all 
    Building configuration: mbed-os-example-client - mbed-os-example-client 
    Updating build tree... 
     
    2  file(s) deleted. 
    Updating build tree... 
    ...
    Total number of errors: 0 
    Total number of warnings: 27 
7) Download and debug => Go
    Starting mbed Client example
    [EasyConnect] IPv4 mode
    [EasyConnect] Using WiFi (ESP8266)
    [EasyConnect] Connecting to WiFi Anfield-tower
    [EasyConnect] MAC address 5c:cf:7f:04:6a:f2
    [EasyConnect] Connection to Network Failed -3012!

    Connection to Network Failed - exiting application...
8) upload .bin made with GCC_ARM => restart K64F
    Starting mbed Client example
    [EasyConnect] IPv4 mode
    [EasyConnect] Using WiFi (ESP8266)
    [EasyConnect] Connecting to WiFi Anfield-tower
    [EasyConnect] Connected to Network successfully
    [EasyConnect] MAC address 5c:cf:7f:04:6a:f2
    [EasyConnect] IP address 192.168.43.142

    SOCKET_MODE : TCP
    Connecting to coap://api.connector.mbed.com:5684

    Registered object successfully!
----------------------------------------------------------------
## Enhancement

**Reason to enhance/problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

teetak01, 2017-12-14:
Note, it is not actually a Client issue if the network-bringup already fails. Should be reported to ESP8266 repository or to Mbed OS instead.

MarceloSalazar, 2017-12-14:
From the user point of view, this application is failing when using ESP8266 and IAR.
@adbridge we should consider adding this to the known issue list.

adbridge, 2017-12-14:
Is this a problem using the IAR exporter or native IAR compilation ?

MarceloSalazar, 2017-12-14:
It seems he is exporting to IAR IDE.
@SamiLappalainenArm could you please compile from the command line and confirm the behavior?
For example:
```
mbed import https://github.com/ARMmbed/mbed-os-example-client
cd mbed-os-example-client
mbed update mbed-os-5.7.0-oob 
mbed compile -t IAR -m K64F -c
```
This will help us to narrow down the problem. Thanks!

SamiLappalainenArm, 2017-12-14:
I did used IAR Embedded Workbecnh IDE when I creted this ticket. 

And when I used "mbed compile -t IAR -m K64F -c" command I got error "Could not find executable for IAR."

I assume that my system configuration is not correct => I will try to fix my system tomorrow.

MarceloSalazar, 2017-12-14:
> "Could not find executable for IAR."
I suspect you'll need to add IAR tools to the system path. You should be able to launch `iccarm` from the command line.

teetak01, 2017-12-15:
I verified that this works fine without online compiler.

SamiLappalainenArm, 2017-12-15:
I managed to use IAR compiler via CLI and here is the client output:

Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (ESP8266)
[EasyConnect] Connecting to WiFi Anfield-tower
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address 5c:cf:7f:04:6a:f2
[EasyConnect] IP address 192.168.43.142

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684

Registered object successfully!
handle_timer_click, new value of counter is 1

MarceloSalazar, 2017-12-15:
@adbridge lets add this to the known issue list when using IAR IDE

closed 2018-03-09 by teetak01