ARMmbed/mbed-os-example-client #357: client fails to connect via UDP when binary is compiled by Online Compiler

SamiLappalainenArm, 2017-12-14:
using UDP fails when binary is compiled with Online compiler but if I compiled with CLI GCC_ARM then UDP works.

---------------------------------------------------------------
## Bug

**Version information**
mbed-os-example-client, revision 9468fd3, tag mbed-os-5.7.0-oob

**Environment details**
https://os.mbed.com/compiler 

**Expected Behavior**
Registered object successfully!

**Actual Behavior**
Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (ESP8266)
[EasyConnect] Connecting to WiFi Anfield-tower
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address 5c:cf:7f:04:6a:f2
[EasyConnect] IP address 192.168.43.142

SOCKET_MODE : UDP
Connecting to coap://api.connector.mbed.com:5684
handle_timer_tick, device not registered

**Steps to Reproduce**
1) compile using TCP => restart K64F
Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (ESP8266)
[EasyConnect] Connecting to WiFi Anfield-tower
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address 5c:cf:7f:04:6a:f2
[EasyConnect] IP address 192.168.43.142

SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684

Registered object successfully!
handle_timer_click, new value of counter is 1

2) change to UDP 
replace "M2MInterface::BindingMode SOCKET_MODE = M2MInterface::TCP;" with
"M2MInterface::BindingMode SOCKET_MODE = M2MInterface::UDP;" in simpleclient.h

3) compile using UDP => restart K64F
Starting mbed Client example
[EasyConnect] IPv4 mode
[EasyConnect] Using WiFi (ESP8266)
[EasyConnect] Connecting to WiFi Anfield-tower
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address 5c:cf:7f:04:6a:f2
[EasyConnect] IP address 192.168.43.142

SOCKET_MODE : UDP
Connecting to coap://api.connector.mbed.com:5684
handle_timer_tick, device not registered

----------------------------------------------------------------
## Enhancement

**Reason to enhance/problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

MarceloSalazar, 2017-12-14:
@JanneKiiskila @adbridge

MarceloSalazar, 2017-12-14:
@SamiLappalainenArm could you please confirm you're switching to the 'mbed-os-5.7.0-oob' branch with the beta Online environment?
https://os-partnerbeta.mbed.com/accounts/login/

SamiLappalainenArm, 2017-12-15:
[image]
I have attached screen capture from Online Compiler revision view => I'm using mbed-os-5.7.0-oob branch from mbed-os-example-client repo

MarceloSalazar, 2017-12-15:
@adbridge let's add this to the known issue list

JanneKiiskila, 2017-12-16:
@SamiLappalainenArm - which version of the CLI compiler did you have?

SamiLappalainenArm, 2017-12-18:
here are details of my CLI compiler:

H:\>armcc.exe
Product: ARM Compiler 5.06
Component: ARM Compiler 5.06 update 6 (build 750)

MarceloSalazar, 2017-12-18:
@theotherjimmy can you please confirm the version of the Arm Compiler being used in the Online IDE?

theotherjimmy, 2017-12-18:
@MarceloSalazar It looks like 5.05, but I would like to have confirmation from @immunda That that is indeed what's installed.

sg-, 2017-12-18:
I've compiled the latest version of Mbed OS and here is what I get.
```
#include "mbed.h"

DigitalOut led1(LED1);

int main() {
    while (true) {
        led1 = !led1;
        wait(0.5);
        printf ("%d\n", __ARMCC_VERSION);
    }
}
```
the output is:
```
5060061
5060061
5060061
5060061
5060061
5060061
```

closed 2018-03-09 by teetak01