ARMmbed/mbed-os-example-client #358: Fails to compile with ARM compiler

vijhar01, 2017-12-14:
This was with 5.7 OOB RC1
Tried the mbed-os-example-client with ARM, GCC, IAR compilers. It failed with ARM and succeeded with others. Tried both standalone and Keil ARM. Same error with both (below). 
- ARM compiler version is 5.06 update 6
- Got it from here - https://developer.arm.com/products/software-development-tools/compilers/arm-compiler/downloads/version-5#panel1a

PS C:\mbed-os-example-client> mbed compile -m K64F -t ARM -c --app-config ./configs/eth_v4.json
Building project mbed-os-example-client (K64F, ARM)
Scan: .
Scan: env
...
....
....
Compile [ 99.8%]: pal_plat_rtos.cpp
[Warning] pal_plat_rtos.cpp@251,0:  #177-D: variable "platStatus"  was declared but never referenced
[Warning] pal_plat_rtos.cpp@252,0:  #177-D: variable "actualOutputLen"  was declared but never referenced
[Warning] pal_plat_rtos.cpp@44,0:  #550-D: variable "g_randInitiated"  was set but never used
Compile [100.0%]: pal_plat_network.cpp
[Warning] pal_plat_network.cpp@589,0:  #1361-D: function "Socket::attach(mbed::Callback<void ()>)"  (declared at line 18
5 of "./mbed-os/features/netsocket/Socket.h") was declared "deprecated"
[Warning] pal_plat_network.cpp@614,0:  #1361-D: function "Socket::attach(mbed::Callback<void ()>)"  (declared at line 18
5 of "./mbed-os/features/netsocket/Socket.h") was declared "deprecated"
[Warning] pal_plat_network.cpp@729,0:  #1361-D: function "Socket::attach(mbed::Callback<void ()>)"  (declared at line 18
5 of "./mbed-os/features/netsocket/Socket.h") was declared "deprecated"
Link: mbed-os-example-client
"C:\Users\xxxxxx\AppData\Local\Temp\p308c-3", line 124 (column 3): Warning: L6312W: Empty Execution region description
 for region RW_IRAM1
Fatal error: L6050U: The code size of this image (256036 bytes) exceeds the maximum allowed for this version of the link
er.
Finished: 0 information, 1 warning, 0 error and 1 fatal error messages.
[ERROR] "C:\Users\xxxxxx\AppData\Local\Temp\p308c-3", line 124 (column 3): Warning: L6312W: Empty Execution region des
cription for region RW_IRAM1
Fatal error: L6050U: The code size of this image (256036 bytes) exceeds the maximum allowed for this version of the link
er.
Finished: 0 information, 1 warning, 0 error and 1 fatal error messages.

[mbed] ERROR: "C:\Python27\python.exe" returned error code 1.
[mbed] ERROR: Command "C:\Python27\python.exe -u C:\mbed-os-example-client\mbed-os\tools\make.py -t ARM -m K64F --source
 . --build .\BUILD\K64F\ARM -c --app-config ./configs/eth_v4.json" in "C:\mbed-os-example-client"
---
PS C:\mbed-os-example-client>

MarceloSalazar, 2017-12-14:
hi @vijhar01 

> Fatal error: L6050U: The code size of this image (256036 bytes) exceeds the maximum allowed for this version of the link

This error suggests that you need to have a license installed. 
Can you please double check?

vijhar01, 2017-12-14:
Yes - I have license installed. 

C:\Users\xxxxx>echo %ARMLMD_LICENSE_FILE%
7010@nahpc-lic19.nahpc.arm.com

C:\Users\xxxxx>

Same error with standalone ARM compiler as well

MarceloSalazar, 2017-12-14:
@vijhar01 it's definitely something wrong with the license, which is not a problem on this actual example app. Maybe you can discuss this with @bridadan in your office?

vijhar01, 2017-12-14:
Will do. 
Planning to try in RC2 and ping Brian D. Thanks

vijhar01, 2017-12-14:
It works now. 
Here are the changes I made

- Used RC2 example
- Uninstalled existing standalone ARM and Keil installs
- Downloaded Keil from \\venus.cambridge.arm.com\Groups\DevSys\mbed\Tools\toolchains\MDK (the .exe with the most recent date - MDK524a)
- Added system env variable ARM_PATH to point to ..Keil_v5\ARM\ARMCC
- Added system env variable ARMLMD_LICENSE_FILE

It compiled and device got registered and working.

Closing this issue.

closed 2017-12-14 by vijhar01