ARMmbed/mbed-os-example-client #359: Client app + LwIP IPv6 fails to connect to network

MarceloSalazar, 2017-12-18:
### Description
- Type: Bug

Tested https://github.com/ARMmbed/mbed-os-example-client
Tag: ree/mbed-os-5.7.0-rc3
That includes fix: https://github.com/ARMmbed/mbed-os/pull/5704
However, this doesn't seem to change the behaviour.

**Expected Behavior**

```
Starting mbed Client example
[EasyConnect] IPv6 mode
[EasyConnect] Using Ethernet
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address b6:47:c9:57:02:cb
[EasyConnect] IP address 2001:470:1D4C:1:B447:C9FF:FE57:2CB
SOCKET_MODE : TCP
Connecting to coap://api.connector.mbed.com:5684
```

**Actual Behavior**

```
Starting mbed Client example
[EasyConnect] IPv6 mode
[EasyConnect] Using Ethernet
<stuck here>
```

**Steps to Reproduce**

```
mbed import mbed-os-example-client
cd mbed-os-example-client/
mbed update mbed-os-5.7.0-rc3
<  update security.h >
copy configs\eth_v6.json mbed_app.json
copy configs\eth-wifi-mbedignore .mbedignore
mbed compile -t ARM -m K64F -c -f

```
@adbridge @mikaleppanen @JanneKiiskila @sg-

teetak01, 2017-12-19:
Known issue on ARM Compiler Update 3. Fixed in ARM Compiler Update 6.

teetak01, 2017-12-19:
https://github.com/ARMmbed/mbed-os/issues/4433

JanneKiiskila, 2017-12-19:
Closing, we can't fix this in the client side. Mbed OS 5.7 should have a patch for it, but the ONLY *real* solution for anyone should be to update their compiler to 5.06 update 6.

mikaleppanen, 2017-12-19:
I cannot reproduce this. When testing with a build made according to steps in "Steps to Reproduce" K64F connects to ipv6 correctly. There should not be any timing/ipv6 negotiation use cases that could differ between ipv6 Ethernet networks.

MarceloSalazar, 2017-12-19:
There seems to be a timing issue involved here. It fails almost always when Armc5u3, but in some few attempts works.
Find here the binaries in case you want to try on your end.

[doesnt-work-armc5u3.zip](https://github.com/ARMmbed/mbed-os-example-client/files/1571471/doesnt-work-armc5u3.zip)
[works-armc5u6.zip](https://github.com/ARMmbed/mbed-os-example-client/files/1571472/works-armc5u6.zip)

JanneKiiskila, 2017-12-19:
Hmmh.. the bug that was in the compiler wasn't timing dependent, it clearly produced 100% faulty code (CRC calculation failed, with 100% rate). This is something else then now.

mikaleppanen, 2017-12-19:
I tried the "doesnt-work-armc5u3.zip" binary in both ipv6 test networks here and it connects always correctly. I cannot repeat the problem here, Marcelo, is it possible for you to trace Ethernet interface (multicasts)? I also made build with some extra lwip ipv6 traces:

[ipv6_traces_armc5u3.zip](https://github.com/ARMmbed/mbed-os-example-client/files/1571679/ipv6_traces_armc5u3.zip)

If there is still problem with ipv6 multicasts, it should not trace any ipv6 input messages.

MarceloSalazar, 2017-12-19:
I've tried the binary with traces enabled - see logs attached: [ipv6_traces_armc5u3_logs.txt](https://github.com/ARMmbed/mbed-os-example-client/files/1571973/ipv6_traces_armc5u3_logs.txt)

However, I'm also seeing this network issue sometimes with GCC and ARMC5u6. I'm wondering if my (tunnel) IPv6 infrastructure is related to this.

JanneKiiskila, 2017-12-20:
Sounds like that.

mikaleppanen, 2017-12-20:
Converted the traces to wireshark log. The router solicitation from k64f is not answered by the router with router advertisement:

[image]

If this happens with all configurations then it must be some issue with network infrastructure.

JanneKiiskila, 2017-12-29:
I would conclude it's an issue with IPv6 tunnel you're using @MarceloSalazar , would you agree? (And close this case).

MarceloSalazar, 2017-12-29:
Yes, it looks like an issue with the tunnel or setup combined with border router.
Closing until we're able to reproduce this consistently.

closed 2017-12-29 by MarceloSalazar