ARMmbed/mbed-os-example-client #364: Latest mbed-os-example-client uses non-official version of mbed Client and libraries

screamerbg, 2017-12-21:
While using mbed CLI 1.3.0 (latest at the time of this writing), I've noticed that newer version of this example program uses non-official version of mbed Client and other libraries.

For example mbed-os-example-client in mbed-os-5.6.1 tag/release uses mbed Client version 3.0.0
```
/tmp/mbed-os-example-client:$ mbed ls
mbed-os-example-client (#19c11e081edf, tags: mbed-os-5.6.1)
|- easy-connect (#bf821b0695cc)
|  |- atmel-rf-driver (#3cc6010188c0)
|  |- esp8266-driver (#fdb44a4bedbb)
|  |  `- ESP8266/ATParser (#a2cb09990f11)
|  |- mcr20a-rf-driver (#9aac10474702)
|  `- stm-spirit1-rf-driver (#bee61bbe6807)
|- mbed-client (#31e5ce203cc0, tags: v3.0.0) # version mbed-client v3.0.0
|  |- mbed-client-c (#ecfa619e42b2, tags: v6.0.0) # version mbed-client-c v6.0.0
|  |- mbed-client-classic (#4e66929607c3)
|  `- mbed-client-mbed-tls (#7e1b6d815038, tags: mbedCloudClient-R1.1, mbedCloudClient-R1.1-RC1)
|- mbed-os (#cc7556a92fb9, tags: mbed-os-5.6.1, mbed_lib_rev152)
`- pal (#60ce64d5ec35)
```

mbed-os-example-client in the latest/mbed-os-5.7.0 release uses unknown version of mbed Client
```
/tmp/mbed-os-example-client:$ mbed update mbed-os-5.7.0
[mbed] Updating program "mbed-os-example-client" to branch mbed-os-5.7.0
[mbed] Updating library "easy-connect" to rev #6af200df4817
[mbed] Updating library "easy-connect/atmel-rf-driver" to rev #2cd9abba3e37
[mbed] Updating library "easy-connect/esp8266-driver" to rev #b0d79dad507d
[mbed] Removing library "easy-connect/esp8266-driver/ESP8266/ATParser" (obsolete)
[mbed] Updating library "easy-connect/mcr20a-rf-driver" to rev #9aac10474702
[mbed] Updating library "easy-connect/stm-spirit1-rf-driver" to rev #0ff4ca7537f0
[mbed] Adding library "easy-connect/wifi-x-nucleo-idw01m1" from "https://github.com/ARMmbed/wifi-x-nucleo-idw01m1" at rev #537b01c4a053
[mbed] Updating library "mbed-client" to rev #ea04c5de7822
[mbed] Updating library "mbed-client/mbed-client-c" to rev #ecfa619e42b2
[mbed] Updating library "mbed-client/mbed-client-classic" to rev #4e66929607c3
[mbed] Updating library "mbed-client/mbed-client-mbed-tls" to rev #7e1b6d815038
[mbed] Updating library "mbed-os" to rev #8b54959f5bca
[mbed] Updating library "pal" to rev #60ce64d5ec35

/tmp/mbed-os-example-client:$ mbed ls
mbed-os-example-client (#5767847be0d2, tags: mbed-os-5.7.0)
|- easy-connect (#6af200df4817)
|  |- atmel-rf-driver (#2cd9abba3e37)
|  |- esp8266-driver (#b0d79dad507d)
|  |- mcr20a-rf-driver (#9aac10474702)
|  |- stm-spirit1-rf-driver (#0ff4ca7537f0)
|  `- wifi-x-nucleo-idw01m1 (#537b01c4a053)
|- mbed-client (#ea04c5de7822)
|  |- mbed-client-c (#ecfa619e42b2, tags: v6.0.0)
|  |- mbed-client-classic (#4e66929607c3)
|  `- mbed-client-mbed-tls (#7e1b6d815038, tags: mbedCloudClient-R1.1, mbedCloudClient-R1.1-RC1)
|- mbed-os (#8b54959f5bca, tags: latest, mbed-os-5.7.0)
`- pal (#60ce64d5ec35)
```

Also looking at the releases available for all libraries, the latest mbed-os-example-client doesn't use release version of the following libraries:
* atmel-rf-driver
* esp8266-driver
* mbed-client
* mbed-client-mbed-tls 

Full log is available below.

Can someone add release tags to there libraries/repositories? CC @JanneKiiskila @yogpan01 @MarceloSalazar 

```
/tmp/mbed-os-example-client:$ mbed releases -r
mbed-os-example-client (#5767847be0d2, tags: mbed-os-5.7.0)
| * mbed-os-5.1.0 
| * mbed-os-5.1.1 
| * mbed-os-5.1.2 
[... 339 lines cut ...]
|    * mbed-os-5.4.2 
|    * mbed-os-5.4.3 
|    * mbed-os-5.4.4 
|    * mbed-os-5.4.5 
|    * mbed-os-5.4.6 
|    * mbed-os-5.4.7 
|    * mbed-os-5.5.0 
|    * mbed-os-5.5.1 
|    * mbed-os-5.5.2 
|    * mbed-os-5.5.3 
|    * mbed-os-5.5.4 
|    * mbed-os-5.5.5 
|    * mbed-os-5.5.6 
|    * mbed-os-5.5.7 
|    * mbed-os-5.6.0 
|    * mbed-os-5.6.1 
|    * mbed-os-5.6.2 
|    * mbed-os-5.6.3 
|    * mbed-os-5.6.4 
|    * mbed-os-5.6.5 
|    * mbed-os-5.6.6 
|    * mbed-os-5.7.0  <- current
`- pal (#60ce64d5ec35)
     No release tags detected
```

JanneKiiskila, 2017-12-22:
Mbed tool chain doesn't support versions, all the hashes are only hashes. Therefore, the versions are more or less irrelevant. Closing out, won't be fixed.

screamerbg, 2017-12-22:
@JanneKiiskila, the mbed toolchain **DOES** support versions. This applies for mbed OS and all example programs, including mbed CLI - see log above. If you're referring to mbed Client lacking versioning after some point then that exactly what I'm asking to be fixed.

JanneKiiskila, 2017-12-27:
Hmm... I wonder why I've not ever seen a single mbed-OS release update .lib change come in with a version number, they are always with hashes? Why is that then?

JanneKiiskila, 2017-12-27:
`mbed releases` doesn't seem to work at least on Windows, fails for me at least. I raised an issue on that.
https://github.com/ARMmbed/mbed-cli/issues/594

JanneKiiskila, 2017-12-29:
So, it is **new** tool feature for mbed-cli, everyone should update their mbed-cli's then...

JanneKiiskila, 2017-12-29:
Why is the tool listing *all* versions for the components? The list is too long.. I would understand if you used something like --all to get ALL the versions, but for now - it should just list ONE version (the one that's used) OR the hash (if no version match).

teetak01, 2018-03-09:
Won't fix.

closed 2018-03-09 by teetak01