ARMmbed/mbed-os-example-client #371: Incorrect SOCKET_MODE set when using mesh config

bentcooke, 2018-01-11:
### Description
- Type: Bug

SOCKET_MODE is incorrectly set to TCP when using the mesh configs causing connections to the connector server to fail. 
---------------------------------------------------------------
## Bug

**Version information**
(`mbed ls`)

**Environment details**
K64F + Atmel 6lowpan shield

**Expected Behavior**
Trace output:
> Starting mbed Client example
> [EasyConnect] IPv6 mode
> [EasyConnect] Using Mesh (Atmel)
> [EasyConnect] Connecting to Mesh...
> [EasyConnect] Connected to Network successfully
> [EasyConnect] MAC address fc:c2:3d:00:00:04:e9:85
> [EasyConnect] IP address 2600:380:607a:14c2:640a:299c:f969:df03
> 
> **SOCKET_MODE : UDP**
> Connecting to coaps://[2607:f0d0:2601:52::20]:5684
> 
> Registered object successfully!
> handle_timer_click, new value of counter is 1
> handle_timer_click, new value of counter is 2

**Actual Behavior**
Trace output:
> Starting mbed Client example
> [EasyConnect] IPv6 mode
> [EasyConnect] Using Mesh (Atmel)
> [EasyConnect] Connecting to Mesh...
> [EasyConnect] Connected to Network successfully
> [EasyConnect] MAC address fc:c2:3d:00:00:04:e9:85
> [EasyConnect] IP address 2600:380:607a:14c2:44e7:eb8d:c43a:7de4
> 
> **SOCKET_MODE : TCP**
> Connecting to coaps://[2607:f0d0:2601:52::20]:5684
> handle_timer_tick, device not registered
> handle_timer_tick, device not registered
> handle_timer_tick, device not registered
> handle_timer_tick, device not registered
> handle_timer_tick, device not registered
> handle_timer_tick, device not registered
> 
> [ERROR:] M2MInterface::NetworkError
> handle_timer_tick, device not registered

**Steps to Reproduce**
Build with any of the mesh configs.

**Potential Fix**
It appears that the issue is due to the order of evaluation of the header files in "main.cpp".    
simpleclient.h evaluates `#if defined (EASY_CONNECT_MESH)` before it gets defined in easyconnect.h.  This can be fixed by simply moving `#include "simpleclient.h"` below `#include "easy-connect/easy-connect.h"` in main.cpp.

teetak01, 2018-01-16:
Thanks for reporting. I'll check this.

teetak01, 2018-01-16:
@bentcooke linked PR to fix this issue based on your suggestions.

PR #374 (2018-01-16): IOTCLT-2328: Fix include ordering
Fixed wrong ordering. Now Mesh connects in correct mode. Fixes #371 

[EasyConnect] Connected to Network successfully
SOCKET_MODE : UDP
Connecting to coaps://[2607:f0d0:2601:52::20]:5684
closed 2018-01-16 by teetak01