ARMmbed/mbed-os-example-client #389: IAR make build fails with K66F

terhei, 2018-03-08:
Exported IAR make build fails with K66F with following error:

"link: mbed-os-example-client.elf"
process_begin: CreateProcess(C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.5\arm\bin\ilinkarm.exe, ilinkarm --skip_dynamic_initialization --threaded_lib --config .././mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/device/TOOLCHAIN_IAR/MK66FN2M0xxx18.icf --output mbed-os-example-client.elf _removed a lot of .o - files here_ )failed.
make (e=87): The parameter is incorrect.
make[1]: *** [mbed-os-example-client.elf] Error 87
make: *** [all] Error 2

artokin, 2018-03-08:
This is similar as https://github.com/ARMmbed/mbed-os-example-mesh-minimal/issues/175.

yogpan01, 2018-03-09:
This is not an application bug but an issue in exporter. Raise it to the right forum for it to get fixed, no action will be taken on this here.

closed 2018-03-09 by yogpan01

adbridge, 2018-03-09:
@terhei @artokin Can one of you please raise this against mbed-os

artokin, 2018-03-12:
This is duplicate of https://github.com/ARMmbed/mbed-os-example-mesh-minimal/issues/175. Issue will be processed there.