ARMmbed/mbed-os-example-client #391: Odin related link section in readme is missing

vijhar01, 2018-03-16:
Couple of issues (both odin_w2 wifi board related)

- This link is not there - https://github.com/ARMmbed/mbed-os-example-client#compile-configuration-for-odin-wifi
- On the 'Testing the Application' section, there are no SW2 or SW3 switches for odin-w2 boards.

ashok-rao, 2018-03-19:
@yogpan01 .. bump!

JanneKiiskila, 2018-03-19:
The issue is a bit vague, can you please extrapolate a bit? 
- Odin specific note about the Mbed OS 5.7.5-5.7.x has now been added (though link is broken, will fix that).
- we should not be referring to SWx anymore, there are the BUTTON1 etc., as an abstraction layer.
  - however, it might be that they do not work unless you pull them up!

ashok-rao, 2018-03-21:
@vijhar01 can you please explain a bit more? Thanks.

vijhar01, 2018-03-21:
The issues I had itemized were my feedback based on 5.8 OOB testing of this example with odin-w2. In this context, 

- Link  https://github.com/ARMmbed/mbed-os-example-client#compile-configuration-for-odin-wifi is broken in the documentation. 
- Please document how to test and monitor application w.r.t odin boards (since they do not have the SW2, SW3 switches). 

I hope this explains. The example itself compiles and works without any issues. No problems with that.