ARMmbed/mbed-os-example-client #394: Without Ethernet : mbed-os-example-client using LPC1347

prvthkr, 2018-04-05:
### Description
- Type: Question
- Priority: Medium
---------------------------------------------------------------
## Question
**Target**
LPC1347

**Toolchain**
ARM

Ethernet port is not available on this target. External [Ethernet shield](https://os.mbed.com/components/W5500-Ethernet-Kit-for-IoT/)(SPI interface) to be used.
    - Can anyone confirm me that this is possible or not?

prvthkr, 2018-04-05:
@teetak01 ??

0xc0170, 2018-04-05:
@JanneKiiskila Do you have any experience with spi ethernet shields running client example?

JanneKiiskila, 2018-04-05:
If you find a working driver for in on Mbed OS - then potentially yes.
However, we have not tested it in any manner. Please bear in mind [board requirements](https://cloud.mbed.com/docs/current/cloud-requirements/production-reqs.html) otherwise.
- TRNG
- enough RAM and Flash.

prvthkr, 2018-04-06:
@JanneKiiskila, Thank you for your information 👍

JanneKiiskila, 2018-04-06:
Hi @prvthkr 

are we OK to close this ticket now, or is there something more?